Vector-Wangel/XLeRobot #4: 腕式相机选择

lovelyyoshino, 2025-05-26:
https://item.taobao.com/item.htm?detail_redpacket_pop=true&id=590682120464&ltk2=1748252449550s82vynhzgne2jo0wcf9qxv&ns=1&priceTId=2150455c17482524391656962e1de3&query=32%20%E6%91%84%E5%83%8F%E5%A4%B4%20usb&skuId=4053819464003&spm=a21n57.1.hoverItem.4&utparam=%7B%22aplus_abtest%22%3A%22016872cd76ccc8745873b3a4f4b2f66b%22%7D&xxc=ad_ztc 是否可以选择这个相机作为腕式相机？

Vector-Wangel, 2025-05-26:
抱歉国内的型号我不太了解，可以自行对比一下参数和尺寸，主要是要适配树莓派。可以买回来自己先试试看，大不了退货😂。

Vector-Wangel, 2025-05-26:
I will close this issue when anyone find a feasible taobao link in China. Feel free to comment and show your results!
我暂时不会关这个issue，如果任何人找到了并验证可用了合适的国内淘宝链接请直接评论谢谢！

lovelyyoshino, 2025-05-27:
已经采购，等待验证

Vector-Wangel, 2025-06-04:
> 想请问下作者腕式相机的安装位置是哪里，没在Assembly文件里看到，是在Lebot机械臂上安装的那个吗

[image]