Vector-Wangel/XLeRobot #16: Thought process of using RealSense D415

ryuzio, 2025-07-17:
Hi,

Thanks for the great work! It is a very interesting project with practical uses, which motivates me a lot to participate.

I see that the latest version (0.2.0) includes an Intel RealSense D415 camera. I wonder if you could share a bit more on the thought process of this selection. (Because it's expensive! And its has very specific functionality limited to computer vision, unlike the power bank.)

To my understanding, the robot needs to gain vision for navigation as well as arm operation. There's already a camera attached to each arm, so the vision for arm operation is somehow covered. But it might be helpful to have a 3rd eye, especially with depth information, to facilitate with the task. That's where D415 could come into the scene IMHO. However, for navigation, I'm not sure if D415 makes the best option comparing to other solutions such as a fish eye camera, LIDAR, etc.

I guess it will be very gauged by the model for controlling the robot. Its capability of process data will limit the types of potential input data, and thus determines the sensors to use.

Vector-Wangel, 2025-07-17:
You are right. But I actually would prefer directly buying a dual-eye camera instead of buying 2 webcam lol. It's much cheaper than Realsense but also have python sdk supports.

Vector-Wangel, 2025-07-17:
> Thanks for the elaboration! Yeah, I'll probably stay with the webcam build for now, and see if it is sufficient for navigation and simple tasks.
> 
> Speaking about being cost-effective, I think maybe 2 webcams is also an option, as I assume that the depth map information could be obtained from 2 stereo images with some algorithm. I'll try and see.

I will add a new head camera connector today so it can connect to both webcams and Realsense cameras.