Vector-Wangel/XLeRobot #17: a problem about realsense

aiforworlds, 2025-07-24:
Thank you for sharing such an excellent open-source project. It's really great.

During the process of trying, I encountered some problems.
i have buy realsense canmera but when i test camera used https://huggingface.co/docs/lerobot/main/en/cameras?shell_restart=Intel+Realsense+Camera  code , i got a error,

lerobot/src/lerobot/cameras/realsense/camera_realsense.py", line 169, in connect
    self.rs_pipeline = rs.pipeline()
                       ^^
NameError: name 'rs' is not defined

error occur  at  here  " import pyrealsense2 as rs", the reason of this error is pyrealsense2  not installed,
I  read pyproject.toml find

 intelrealsense = [
    "pyrealsense2>=2.55.1.6486 ; sys_platform != 'darwin'",
    "pyrealsense2-macosx>=2.54 ; sys_platform == 'darwin'",
]

The support for the Raspberry Pi (ARM) platform by pyrealsense2 is limited.

so i want to know how to install pyrealsense2 in Raspberry Pi，and which version have you installed，
thands
best regards

Vector-Wangel, 2025-07-24:
We also tested that raspberry pi might not support Realsense directly. So I suggest you also try webcam as head camera first. And in the near future we will offer support for NVIDIA jetson as  one our upgraded computing platforms.

aiforworlds, 2025-07-29:
我已经测试成功，在树莓派上手动编译安装Realsense，可以成功输出rgb和depth图像，具体的安装步骤如下：
1）下载最新版https://github.com/IntelRealSense/librealsense
2）临时强制 Python 进程加载系统新版 libstdc++
  export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libstdc++.so.6
3）sudo apt update
  sudo apt install -y \
      libx11-dev \
      libxrandr-dev \
      libxinerama-dev \
      libxcursor-dev \
      libxi-dev \
      libgl1-mesa-dev \
      libglu1-mesa-dev \
      libvulkan-dev \
      libusb-1.0-0-dev \
      libgtk-3-dev \
      pkg-config
  sudo apt install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev cmake
  mkdir build && cd build
  cmake .. -DBUILD_PYTHON_BINDINGS=ON -DBUILD_WITH_OPENGL=OFF
  make -j4
  sudo make install

5)安装成功后使用命令python -m lerobot.find_cameras realsense查看摄像头的id然后
把id填入下面的serial_number_or_name中

from PIL import Image
import numpy as np
from lerobot.cameras.realsense.configuration_realsense import RealSenseCameraConfig
from lerobot.cameras.realsense.camera_realsense import RealSenseCamera
from lerobot.cameras.configs import ColorMode, Cv2Rotation

# 配置摄像头
config = RealSenseCameraConfig(
    serial_number_or_name="134322073242",  # 替换为你的摄像头序列号
    fps=15,
    width=640,
    height=480,
    color_mode=ColorMode.RGB,
    use_depth=True,
    rotation=Cv2Rotation.NO_ROTATION
)

camera = RealSenseCamera(config)
camera.connect()

try:
    # 获取彩色图和深度图
    color_frame = camera.read()  # 形状 (H, W, 3), dtype=uint8, RGB格式
    depth_map = camera.read_depth()  # 形状 (H, W), dtype=float32

    # 保存彩色图 (Pillow 直接支持 RGB)
    color_image = Image.fromarray(color_frame)
    color_image.save("color_image_pillow.png")

    # 保存深度图 (归一化到0-65535并转为16位)
    depth_normalized = (depth_map * (65535.0 / depth_map.max())).astype(np.uint16)
    depth_image = Image.fromarray(depth_normalized)
    depth_image.save("depth_image_pillow.png")

    print("图像保存成功！")
    print(f"彩色图路径: color_image_pillow.png")
    print(f"深度图路径: depth_image_pillow.png")

finally:
    camera.disconnect()

Vector-Wangel, 2025-08-01:
我这边目前准备暂时先不做realsense转而先做双目相机😂，所以可以看看其他人有没有能解决这个问题的。关于另一个问题，建议直接去lerobot或lekiwi也问问，我短期内应该也会专注于先在笔记本上实现xlerobot。

hyy02, 2025-08-19:
@aiforworlds 我觉得可以考虑买一个树莓派的USB免驱扬声器，成本很低大概30元左右，或者稍微改下代码显示出记录的开始和截止~

closed 2026-01-18 by Vector-Wangel