Vector-Wangel/XLeRobot #30: I did not find the fetch on the github web

beforeface, 2025-09-01:
[image]

[image]
How could I replace these files ? I did not find these files

hyy02, 2025-09-02:
1. For the question 1, I try the author's code, it's just a warning, you can change the [code](https://github.com/Vector-Wangel/XLeRobot/blob/358f536ef6cbe0cf4203ff427ffe5fb4056885ba/simulation/Maniskill/envs/scenes/base_env.py#L40) here to `SUPPORTED_ROBOTS = ["panda", "fetch", "xlerobot"]` , and make sure you change the scene_builder and __init__ described in the author's [README](https://github.com/Vector-Wangel/XLeRobot/tree/main/simulation/Maniskill).
2. For the question 2, these are the scripts for real world lerobot control, it's not for the maniskill simulation. If you want to run simulation for it please see the [simulation](https://github.com/Vector-Wangel/XLeRobot/tree/main/simulation/Maniskill) part.
3. For the question 3
```
model -> leorbot/src/lerobot/model
robots/xlerobot ->lerobot/src/lerobot/robots/xlerobot
```
@beforeface

Vector-Wangel, 2025-09-02:
Thanks for the questions and answering, I will update the documentation today to address these concerns.

beforeface, 2025-09-02:
Okay ,thank you.
1. Have you successfully run the double-arm version? I tried to add the "xlerobot" uid, still got a problem
2. I bought some materials from web, meanwhile I tried to understand the tutorial and found the issue.
3. Maybe the files for the real world XLerobot? I need to mkdir the src files?

hyy02, 2025-09-02:
@beforeface 
1. Can you show your terminal output, so I can know where the bug is?
2. Good job, welcome to join the XLerobot family
3. You needn't, you just need to clone the [lerobot](https://github.com/huggingface/lerobot) repo

hyy02, 2025-09-02:
It seems that you choose the wrong controllerm, end effector control need the pd_joint_delta_pos_dual_arm, please use the command in the documentation.
`python -m mani_skill.examples.demo_ctrl_action -e "ReplicaCAD_SceneManipulation-v1"   --render-mode="human" --shader="rt-fast" -c "pd_joint_delta_pos_dual_arm"`

beforeface, 2025-09-02:
@hyy02 No, it still not work for it. the message is "pd_joint_delta_pos_dual_arm not in supported modes". Actually I'm a new guy for the mani-skill, it looks like I need to edit the mani-skill's config?

[image]

hyy02, 2025-09-02:
Sorry :joy: , I forgot to add `-r "xlerobot"`, the default agent setting is 'fetch', I test  this: 
`python -m mani_skill.examples.demo_ctrl_action -e "ReplicaCAD_SceneManipulation-v1"   --render-mode="human" --shader="rt-fast" -c "pd_joint_delta_pos_dual_arm" -r "xlerobot"`

beforeface, 2025-09-02:
@hyy02 Yeah, thank you. The double-arm version successfully run. I have a question: how could you know the -c parameter is wrong?

hyy02, 2025-09-02:
@beforeface Actually, I don't really understand your question 😂 , Just follow the author's documentation, use the controller config he specified.  the single-arm and dual-arm use different controller config, checkout the terminal output with the author's documentation

closed 2026-01-18 by Vector-Wangel