Vector-Wangel/XLeRobot #53: questions about xlerobot

jcl2023, 2025-09-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jcl2023, 2025-09-15:
@Vector-Wangel 

 I plan to buy xlerobot from wowrobo. Before I do that, I have several questions regarding this robot:

1)It mentions realsense camera can be used, which realsense camera and how to mount it to xlerobot?
2)Does sw support training with realsense inputs along with rgb camera? 
3)How to do teleoperation to capture datasets with single and dual arms?

Thanks

hyy02, 2025-09-17:
@jcl2023 Sure, it's not difficult, you can modify the code yourself refer to the repo SW code.

jcl2023, 2025-09-21:
@Vector-Wangel @hyy02 

>>you can also get started with huggingface lerobot first where they have official support for realsense.
I don't see realsense support on huggingface lerobot. Can you point me where the realsense camera support code is?

Thanks