Vector-Wangel/XLeRobot #56: docs: Incorrect command for the VR control demo

tc-huang, 2025-09-16:
**Describe the bug**
The command provided in the documentation for the VR control demo is incorrect. It points to a non-existent Python module and is missing a required argument, which causes a `ModuleNotFoundError` when executed.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the command in your terminal:
    ```bash
    python -m mani_skill.examples.XLeRobot_demo_VR_ee_ctrl \
        -e "ReplicaCAD_SceneManipulation-v1" \
        --render-mode="human" \
        --shader="rt-fast" \
        -c "pd_joint_delta_pos_dual_arm"
    ```
2. See the error: `No module named mani_skill.examples.XLeRobot_demo_VR_ee_ctrl`
 
**Expected behavior**
The command should execute without errors and launch the VR simulation demo successfully.

**Screenshots**
  [image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version 22.04

hyy02, 2025-09-16:
@tc-huang  Please use this one: `python -m mani_skill.examples.demo_ctrl_action -e "ReplicaCAD_SceneManipulation-v1"   --render-mode="human" --shader="rt-fast" -c "pd_joint_delta_pos_dual_arm" -r "xlerobot"`, we'll fix this documentation bug.

hyy02, 2025-09-16:
Oh, sorry 😂 , If you want to use VR control, please use demo_ctrl_action_ee_VR, demo_ctrl_action is for the keyboard.

hyy02, 2025-09-16:
@tc-huang Thanks you very much for improving the project.

closed 2025-09-20 by tc-huang